OctoMap/octomap_mapping #13: Dynamic reconfigure in server

jvgomez, 2015-05-29:
Hi, before doing the job I prefer to ask, just in case:

- Is there any reason why the octomap_server only has the max_depth parameter in the dynamic reconfigure server and not other parameters, such as those for cloud filtering or ground segmentation?

In case there is no inconvenient, I suggest it as an enhancement. I would most probably do it by myself in that case.

ahornung, 2015-05-30:
There's no reason not to have additional parameters in dynamic_reconfigure, except the work required to implement it ;)

Max depth was just the most useful and obvious one to start with. Contributions are welcome!

jvgomez, 2015-06-04:
I have the server running, but not really clean because of this problem: http://answers.ros.org/question/210400/dynamic-reconfigure-and-parameter-namespaces-in-octomap_server/

Do you have any suggestion on how to overcome it?

closed 2015-06-22 by jvgomez