OctoMap/octomap_mapping #30: Is ROS Kinetic supported yet?

sytelus, 2016-07-23:
I don't see octomap_mapping or octomap_server packages in ros-kinetic. Is Kinetic supported yet? These are the only packages that is preventing us to move on to Kinetic :(. It's especially bad because if we want to downgrade to Jade then we also have to downgrade OS to Ubuntu 14.04 because Ubuntu 15 support expires in few month. 

Is there anyway to use mapping and server packages on Kinetic?

Thanks!

ahornung, 2016-07-24:
I currently lack the time to test octomap_server in Kinetic. If there is support from the community by testing the compilation and basic functionlity of octomap_server in Kinetic, then this would drastically speed up a release

Basically, the state of `kinetic-devel` (I just created a new branch based on `jade-devel` should just work.

sytelus, 2016-07-25:
May be we can just start with that? If you can publish these packages in Kinetic release then we can start using them and file bugs as we encounter.

ahornung, 2016-07-29:
A broken package can delay the complete build farm. Only tested and working packages should be released, otherwise users will stumble over non-working code.

But on the other hand there's no need to have a binary release for testing and bug fixing. Why not add octomap_mapping to your catkin source checkout and compile?

LorenzMeier, 2016-08-02:
I've run into this build error on Kinetic, which seems a complete link fail. @tfoote is there a specific Jade -> Kinetic upgrade step / guide required? Have you seen this before?

```sh
CMakeFiles/octomap_server_multilayer.dir/src/octomap_server_multilayer.cpp.o: In function `main':
octomap_server_multilayer.cpp:(.text+0x76): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
octomap_server_multilayer.cpp:(.text+0x1fd): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
octomap_server_multilayer.cpp:(.text+0x311): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
octomap_server_multilayer.cpp:(.text+0x47e): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
octomap_server_multilayer.cpp:(.text+0x7ae): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::setParam(std::string const&, int) const'
devel/lib/liboctomap_server.so: undefined reference to `tf::Transformer::getLatestCommonTime(std::string const&, std::string const&, ros::Time&, std::string*) const'
devel/lib/liboctomap_server.so: undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::getParam(std::string const&, double&) const'
devel/lib/liboctomap_server.so: undefined reference to `tf::Transformer::waitForTransform(std::string const&, std::string const&, ros::Time const&, ros::Duration const&, ros::Duration const&, std::string*) const'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::getParam(std::string const&, bool&) const'
devel/lib/liboctomap_server.so: undefined reference to `tf::Transformer::lookupTransform(std::string const&, std::string const&, ros::Time const&, tf::StampedTransform&) const'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::setParam(std::string const&, double) const'
devel/lib/liboctomap_server.so: undefined reference to `octomap::AbstractOccupancyOcTree::readBinary(std::string const&)'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
devel/lib/liboctomap_server.so: undefined reference to `tf::Transformer::canTransform(std::string const&, std::string const&, ros::Time const&, std::string*) const'
devel/lib/liboctomap_server.so: undefined reference to `tf::TransformListener::transformPoint(std::string const&, geometry_msgs::PointStamped_<std::allocator<void> > const&, geometry_msgs::PointStamped_<std::allocator<void> >&) const'
devel/lib/liboctomap_server.so: undefined reference to `octomap::AbstractOcTree::read(std::string const&)'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::getParam(std::string const&, int&) const'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::setParam(std::string const&, bool) const'
collect2: error: ld returned 1 exit status
make[2]: *** [devel/lib/octomap_server/octomap_server_multilayer] Error 1
make[1]: *** [CMakeFiles/octomap_server_multilayer.dir/all] Error 2
CMakeFiles/octomap_server_node.dir/src/octomap_server_node.cpp.o: In function `main':
octomap_server_node.cpp:(.text+0x76): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
octomap_server_node.cpp:(.text+0xef): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
octomap_server_node.cpp:(.text+0x2ce): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
octomap_server_node.cpp:(.text+0x3e2): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
octomap_server_node.cpp:(.text+0x500): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
octomap_server_node.cpp:(.text+0x5b9): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
octomap_server_node.cpp:(.text+0x76e): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
octomap_server_node.cpp:(.text+0xb63): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::setParam(std::string const&, int) const'
[... cut ...]
devel/lib/liboctomap_server.so: undefined reference to `octomap::AbstractOcTree::read(std::string const&)'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::getParam(std::string const&, int&) const'
devel/lib/liboctomap_server.so: undefined reference to `ros::NodeHandle::setParam(std::string const&, bool) const'
collect2: error: ld returned 1 exit status
make[2]: *** [devel/lib/octomap_server/octomap_server_node] Error 1
make[1]: *** [CMakeFiles/octomap_server_node.dir/all] Error 2
make: *** [all] Error 2
```

james-ward, 2016-09-26:
I have built and deployed the kinetic-devel branch on ROS kinetic. It builds and runs fine.

KvalheimRacing, 2016-10-15:
How did you do it?

james-ward, 2016-10-15:
@KvalheimRacing It's a pretty standard catkin build.
Make a workspace ( http://wiki.ros.org/catkin/Tutorials/create_a_workspace ), then clone the source into it:
```
source /opt/ros/kinetic/setup.bash
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
catkin_init_workspace
git clone -b kinetic-devel https://github.com/OctoMap/octomap_mapping.git
cd ~/catkin_ws
catkin_make
catkin_make install
```

KvalheimRacing, 2016-10-16:
Ah ofc, thanks!

KvalheimRacing, 2016-10-16:
Is there any prefered target link library? In my CMakeLists.txt file, I dont know how to make a library/"MY_TARGET_NAME". Or, I know that the library not should be made in the CMakeList.txt file, but I dont know where.

ahornung, 2016-10-19:
> I have built and deployed the kinetic-devel branch on ROS kinetic. It builds and runs fine.

Thanks! A release for kinetic (0.6.1) is on the way to the build farm.

james-ward, 2016-10-26:
The package appeared on the repo yesterday.
Just installed it - works nicely.

closed 2016-10-27 by ahornung

saurabhbansal90, 2018-04-09:
Hello Guys, I am planning to use Octomap for my robot arm manipulation, for obstacle avoidance.
So for that purpose, I think  I will be following steps below:
1. Get the point cloud from sensor
2. Run a Octomap server node to receive the point cloud
3. From Octomap server, subscribe to the binary octomap

Then how to pass this binary data to some path planner for obstacle avoidance.

Can somebody please confirm, if I have to use this approach only.
And if there is some example implementation  that I can make use of.

With best regards,
Saurabh

ahornung, 2018-04-22:
Hello Saurabh,

your best bet would be to look at the MoveIt! package. It incorporates 
all the steps + manipulation planning.

Best regards,

Armin

saurabhbansal90, 2018-04-23:
Hello Armin, thanks for your response.
Yes its available in MoveIt package only. I folllowed this [link](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/perception_configuration.html).

It is working fine apart from showing a TF error. I am resolving that error.