OctoMap/octomap_mapping #31: Problem with ground filter

dangquang1993, 2016-07-24:
Hi,

## Context
I am using octomap with my quadcopter with three small stereo cameras in an exploration mission where the elevation of the ground is unknown but can vary by sometimes a maximum of 2meters over the course of the entire flight. We use the 2D projected map to do our path planning and we would like to have the ground marked as free space in the projection map. Our problem is that when we put `occupancy_min_z` too high we clear only the space very close to us. However if we put the `occupancy_min_z` too low, tend to have the ground marked as an obstacle.

## Problem
Our idea was to enable ground filtering to clear unknown space into free space faster.

However, when we run the octomap_server, we get multiple PCL errors: 

`[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.`

Here is the launch file we are using:
```xml
<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
    <param name="resolution" value="0.2" />

    <!-- fixed map frame (set to 'map' if SLAM or localization running!) -->
    <param name="frame_id" type="string" value="$(arg frame_id)" />
    <param name="base_frame_id" type="string" value="$(arg base_frame_id)" />

    <!-- maximum range to integrate (speedup!) -->
    <param name="sensor_model/max_range" value="3.0" />

    <!-- data source to integrate (PointCloud2), we modified octomap to support 3 point cloud inputs-->
    <remap from="cloud_in"  to="$(arg points1)" />
    <remap from="cloud_in2" to="$(arg points2)" />
    <remap from="cloud_in3" to="$(arg points4)" />
    <param name="filter_ground" type="bool" value="true" />
    <param name="filter_speckles" type="bool" value="true" />
    <param name="ground_filter/distance" type="double" value="0.2" />
    <param name="ground_filter/plane_distance" type="double" value="0.2" />
    <param name="pointcloud_min_z" type="double" value="-2.0" />
    <param name="pointcloud_max_z" type="double" value="4.0" />
    <param name="occupancy_min_z" type="double" value="1.0" />
    <param name="occupancy_max_z" type="double" value="2.5" />
</node>
```
The error happens when octomap_server is doing ground plane filtering. It occurs only when the filter_ground parameter is enabled. 

## Questions
We have two questions:

- What is causing this PCL filter problem?
- Are we doing this the right way or is the ground filtering unnecessary?

Thanks!

ahornung, 2016-07-29:
That PCL error / warning appears when there can be no plane estimated, usually when there are to few or no points. Maybe the parameter `sensor_model/max_range` is too low?

You could also try debugging with rqt_console, depending on the log level set you'll see exactly what octomap_server is doing.

blup-Blupon, 2017-06-14:
Hi !  I'm having the same issue, with or without defining a `sensor_model/max_range`

For example I run: 

`rosrun octomap_server octomap_server_node cloud_in:=/pcl_xyz _occupancy_min_z:=-.85 _occupancy_max_z:=0.5 _filter_ground:=true _ground_filter/plane_distance:=0.8
`

which includes points belonging to the ground.

To get similar errors:

`[INFO] [1497343346.217816837]: PCL segmentation did not find any plane.
[ WARN] [1497343346.217868956]: No ground plane found in scan
[ERROR] [1497343346.426143499]: Transform error for ground plane filter: "base_footprint" passed to lookupTransform argument target_frame does not exist. , quitting callback.
You need to set the base_frame_id or disable filter_ground.
^C[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.`

The PointCloud2 I'm using as input for octomap_server is published by a [Parrot SlamDunk](https://www.parrot.com/us/business-solutions/parrot-slamdunk) ROS node, it has a reasonable amount of points, although I admit not that much belonging to the ground, and thus, enabling a successful plane detection.

ahornung, 2017-06-14:
If there are too few points for a reasonable ground plane fit then you should not use the ground filter.

However, I'm seeing a `Transform error` in your log. Could it be that you're missing the proper base and base_footprint tf transforms?

blup-Blupon, 2017-06-15:
Indeed ! I just tried it again, this time with proper parameters for [octomap_server](http://wiki.ros.org/octomap_server) **~frame_id** and **~base_frame_id** (`view_frames` pdf return [here](https://nuage.ensta.fr/index.php/s/jQgmts4DOFSWXVz)), getting the no ground plane found warning again.

### Command-line messages:

`[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[ INFO] [1497517506.648828231]: PCL segmentation did not find any plane.
[ WARN] [1497517506.648858067]: No ground plane found in scan
[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.`

### Command-line command:

`rosrun octomap_server octomap_server_node cloud_in:=/pcl_xyz _occupancy_min_z:=-1 _occupancy_max_z:=0.5 _frame_id:=/map _base_frame_id:=/cam0 _filter_ground:=true _ground_filter/plane_distance:=0.7`

I have quite a lot of points in my PointCloud2 (.pcd example file [here](https://nuage.ensta.fr/index.php/s/Sw7m6EA3cYu4rbM)), though not on the ground, I don't know why my sensor doesn't detect much there...

Anyway, if I can't do anything with octomap_server parameters to force the ground filter to work I guess I'll just have to find another way to do it. Thanks for your answer ! :+1:

snakehaihai, 2017-12-05:
I tried octomap output with my own pcl ground filter node. it works fine. So you can use it same way as well

BrunoEduardoCSantos, 2018-07-10:
Did anyone find a solution?

Linearl, 2019-07-17:
just look inside the code and you will find some functions in <octomap_server.cpp>.
in function `OctomapServer::insertCloudCallback`, author use function `void OctomapServer::filterGroundPlane` to do the filter work. The base idea is to find a plane which is vertical to the axis-Z in world cordinate.  zmin zmax are used to decide the inliers, if your drone fly in different altitude， you should modify both your code and octomap_server code rather than enlarge the range between` zmin~zmax`, which will put too many points into account and `pcl::SACSegmentation` may not works . Notify octomap_server the change and it will work properly.

tomkimsour, 2021-05-17:
Did anyone find a solution to this ? Which parameter should i tweak in the ros node to solve this ?
Is it ```pointcloud_max_z``` and ```pointcloud_min_z``` ? What range should I use.
I did not really understand the last answer