OctoMap/octomap_mapping #39: Problem remapping topic using nodelet octomap_server

james-ward, 2016-10-31:
I originally posted a question on ROS Answers, but I've done some more experimenting and I think this is specific to the nodelet implementation of octomap server.

I cannot get a pointcloud topic in the same nodelet manager to remap to the input of the octomap server nodelet.

For example, neither of these lines work:
`<remap from="~cloud_in" to="depth/points" />`
or
`<remap from="octomap_server_nodelet/cloud_in" to="depth/points" />`

When I use the same style of remapping for a PCL nodelet (for example) they work fine and the data is passed through the nodelet manager. So the point cloud source and the nodelet manager are working, which makes me think that this is a problem with the nodelet implementation in octomap. Unfortunately I'm not familiar enough with ROS nodelets to be able to work out the fix.

Has anyone else got a successful octomap nodelet running?

tcassell94, 2018-01-10:
Old issue, but still exists, so...

I believe this is because the octomap server class isn't a native nodelet class (ie, doesn't inherit from nodelet::Nodelet), and therefore doesn't do the "proper" nodelet thing and use a call to `getPrivateNodeHandle()`. It simply gets its own nodehandle. When wrapped by the octomap server nodelet, this nodehandle becomes the nodehandle of the nodelet manager. Therefore, to remap topics, put the remapping arguments in the namespace of the manager, rather than the namespace of the octomap server nodelet. Unsure why this method was chosen, but it can be made to work by using the manager's namespace for remappings.

yputra, 2019-11-03:
Anyone has solved this issue? I used to only use the octomap server node, now when I want to use the nodelet, it has the same issue, as it could not detect the cloud in remap topic. Any suggestion would be helpful!

wxmerkt, 2019-11-04:
@yputra can you please try #42 and feed back whether that resolves the issue for you? If so, I will update that PR, merge it, and make a new release.

yputra, 2019-11-04:
@wxmerkt Hi Wolfgang, yes I have implemented #42 and it is working well for me now. I think you should merge it.

Just a note that after implementing #42 , the remap of the input should be from "cloud_in" instead of "octomap_server_nodelet/cloud_in".

wxmerkt, 2019-11-04:
Thank you very much for the quick feedback. We will have to fix #42 to preserve current behaviour.

wxmerkt, 2019-11-05:
@yputra Can you please try #61?

yputra, 2020-01-08:
@wxmerkt Hi Wolfgang, sorry for the late reply. Yes it is now working as well with #61 . Thanks a lot for the support!

wxmerkt, 2020-01-08:
Thank you for the feedback. This has now been merged to `kinetic-devel`.

closed 2020-01-08 by wxmerkt