OctoMap/octomap_mapping #48: Update Octomap in realtime using octomap_rviz_plugin

BrunoEduardoCSantos, 2018-07-20:
Hi,
This is a question why my octomap_server is overlaying the several octomaps created. 
My hunch is that I am missing the right frame_id or some tf from SLAM. The SLAM algorithm is LOAM which have two frames: camera ; camera_init. From this, I wrote the following launch file:
`
<launch>
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
		<param name="resolution" value="0.05" />
		
		   <param name="frame_id" value="camera_init" />
		<param name="base_frame_id" value="camera" />
		
		<param name="sensor_model/max_range" value="130.0" />
               <param name="latch" value="false" />
	       <param name="filter_speckles" value="true" />
            <param name="pointcloud_min_z" value="-0.6" />
	    <param name="pointcloud_max_z" value="0.5" />
           <param name="occupancy_min_z" value="-0.5" />
            <param name="occupancy_max_z" value="2.5" />
          
	<remap from="cloud_in" to="/velodyne_cloud_registered" />
      </node>
</launch>`

What am I doing wrong?
Thanks,
Bruno

haakonelf, 2019-05-06:
Hey! Did you figure it out? 
Thanks

closed 2019-05-07 by BrunoEduardoCSantos

haakonelf, 2019-05-07:
I will take that as a no?