OctoMap/octomap_mapping #49: Release for ROS Melodic Morenia

mjbogusz, 2018-08-28:
Will `octomap_server` (and `octomap_ros`) be released (packaged and available in repo) for ROS Melodic?

I've tested the `melodic-devel` branch by adding the packages to a catkin workspace and it worked flawlessly as far as our usecase goes (Gazebo simulation, Kinect for mapping and planning robot motion based on current octomap state).

ahornung, 2018-10-04:
I'm not planning any more releases of octomap_server at the moment, since I can't maintain it myself anymore. Even if working "as-is", there are bound to be patches and fixes necessary at some point in time. So I think it's best to add it to a source workspace and build manually.

I added another release of octomap_ros though, even if it doesn't look like there are any packages directly depending on it.

wxmerkt, 2019-01-14:
@ahornung I am happy to volunteer to do the release for melodic and look at pull requests and fixes that may come in, time permitting. There are some upstream dependent packages which would be a great benefit to the community if we could get them released/unblocked (e.g., `jsk_rviz_plugins`).

ahornung, 2019-01-17:
Sounds good to me, thanks for volunteering!

@mikaelarguedas @dirk-thomas @wjwwood could you add @wxmerkt  as maintainer to ros-gbp/octomap_mapping-release?

dirk-thomas, 2019-01-17:
> could you add @wxmerkt as maintainer to ros-gbp/octomap_mapping-release?

Done.

wxmerkt, 2019-01-17:
Thank you very muich @ahornung and @dirk-thomas! I am happy to make the release. Are there any outstanding pull requests or features we'd like to get in before releasing? Otherwise I'll release `0.6.1` as on kinetic to melodic.

wxmerkt, 2019-01-18:
I have released `0.6.1` to melodic from `kinetic-devel`: https://github.com/ros/rosdistro/pull/19985 as there are no changes yet in `melodic-devel`.

ahornung, 2019-01-24:
@wxmerkt I'm getting a few "Build failed" messages from the build farm, could you look into it? It might be best to switch out my email address for yours as maintainer.

wxmerkt, 2019-01-24:
I am happy to look into it. Could you please give me write access to this repo as well?

ahornung, 2019-01-26:
Done, added you to all the ROS repos `octomap_*`.
One of the failing jobs is here: http://build.ros.org/job/Mbin_ubv8_uBv8__octomap_server__ubuntu_bionic_arm64__binary/3/display/redirect

They all appear to be ARM jobs, so maybe they just need to be blacklisted for the platform.

wxmerkt, 2019-01-26:
Thank you! I’ll have a look into it and will update the maintainer email.

wxmerkt, 2019-01-29:
`octomap_mapping` has now been released to `melodic`. It will be available after the next melodic sync.

closed 2019-01-29 by wxmerkt

iscsrl, 2020-05-21:
Hi, any update about the release for Ros Melodic?

wxmerkt, 2020-05-21:
This package has been released and is available on ROS Melodic: `ros-melodic-octomap-mapping`. Please let me know if you have any issues