OctoMap/octomap_mapping #62: Can OctoMap unoccupy?

bomarali, 2019-11-11:
Does OctoMap work with dynamic objects at all? Say if I want to grasp an object and move it, will the OctoMap reflect the change? So far I have been playing with sensor model but I cannot seem to make the OctoMap forget the original position of the object.

ahornung, 2019-11-15:
There is no explicit decay or labeling of dynamic objects. For an object to clear, you need to see a sufficient number of points in the background behind the removed object, or modify the code to use "maxrange" readings for clearing only. A tweaked sensor model helps to clear an object faster.

bomarali, 2019-12-03:
My bad, I transformed the point cloud to a different frame before feeding into the octomap server.

closed 2019-12-03 by bomarali

murikill, 2021-07-08:
I'm still having the same issue - dynamically objects won't removed. I think, there are enough points in the background, but the cells doesn't disappear. What do I wrong? 
Thank you!
[image]

Jarvis-sustech, 2021-11-08:
> I'm still having the same issue - dynamically objects won't removed. I think, there are enough points in the background, but the cells doesn't disappear. What do I wrong? Thank you! [image]

Have you solved it? I am facing the same thing.