OctoMap/octomap_mapping #63: Nothing to publish, octree is empty

kaponloto, 2019-11-13:
I am a publishing a PointCloud (Type: sensor_msgs/PointCloud2) in topic pc_pub. I have remapped "cloud_in" topic to the according pointcloud topic and the fixed frame in the launch file corresponds to the fixed frame in rviz (map) . Running  ran "rostopic echo pc_pub" verifies that the pointcloud is indeed being published to this topic and octomap_server is Subscribing into the topic too. However, "Nothing to publish, octree is empty" appears.
What could the problem be?

<launch>
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
		<param name="resolution" value="0.05" />
		
		
		<param name="frame_id" type="string" value="map" />
		
		
		<param name="sensor_model/max_range" value="100.0" />
		
		
		<remap from="cloud_in" to="/pc_pub" />
	
	</node>
</launch>

borcherd, 2020-03-02:
@kaponloto Did you ever solve it?

wxmerkt, 2020-03-02:
Is the map frame being published?

FeiyangJin, 2020-03-09:
You may need to specify frame_id and base_frame_id to the same as you used in Rviz. In my case, I did it like:
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
		<param name="resolution" value="0.05" />
		
		
		<param name="frame_id" type="string" value="world_enu" />
		
		
		<param name="sensor_model/max_range" value="20.0" />
		
		
		<remap from="cloud_in" to="/airsim_node/drone_1/front_left_custom/DepthPlanner/registered/points" />
		<param name="base_frame_id" value="world_enu" />
	</node>

tomkimsour, 2021-04-30:
How did you fix it ?

leonardlohky, 2021-06-22:
Also facing this same issue under similar circumstances. Has anyone been able to find a solution?

closed 2021-07-29 by agchatzik

sherlockholmes1603, 2022-01-03:
Sir we are getting the same error and also there is warn status by visualizing "/occupied_cells_vis_array" topic in  rviz and also "/free_cells_vis_array" topic is empty while visualizing the topic in rviz

Please help me guide me what to do

tejalbarnwal, 2022-01-28:
Hey, you need to ensure that there is a transformation from the `frame_id` that you had specified in the `octomap_mapping.launch` and the frame_id that is present inside the header part of the sensor data.

ShuiYunXi, 2022-05-06:
I also get the same error and how to  set the transformation  frame_id ?
my error info:
----------------
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [octomap_server_node-1]: process started with pid [11004]
[octomap_server_node-1] [INFO] [1651821775.666505026] [octomap_server]: Publishing latched (single publish will take longer, all topics are prepared)
[octomap_server_node-1] [WARN] [1651821775.689450702] [octomap_server]: Nothing to publish, octree is empty
[octomap_server_node-1] [WARN] [1651821775.689514529] [octomap_server]: Could not open file

pathakarvindk, 2023-01-11:
@ShuiYunXi 
Did you get any solution. My environment is also foxy and ubuntu20.04.
I have point cloud, need to consider that as a collision object in the form of octomap.

Any help is appreciated.

HuaYuXiao, 2024-03-15:
If your map is relatively big, just wait for a while and see what happens.

flilqhu, 2025-03-05:
same issue...my enviroment is foxy and ubuntu20.04