OctoMap/octomap_mapping #65: /octomap_point_cloud_centers topic is not clear when there is a loop closure in rtabmap_ros

acpl00, 2020-02-12:
I have a launch file that contains the following two packages, among others:

 `rtabmap.launch`

 `octomap_server with cloud_in = /rtabmap/cloud_map`

I do have a problem, when I run the `rtabmap.launch`  and I plot the `/octomap_point_cloud_centers` in `RVIZ`.

The problem is that when the rtabmap finds a loop closure and corrects itself, the /octomap_point_cloud_centers topic do not clear.

How can I solve this issue?

Any help is appreciated, thank you

wxmerkt, 2020-02-12:
Hi!
Sorry I am not familiar with the particular launch file, rtabmap, or how it uses the octomap package or which node thereof. You should be able to make a service call to reset the octomap which potentially would clear the point cloud?

acpl00, 2020-02-13:
Hi,
I have called the reset service call, and indeed in resets the `cloud_map`,  but it does not reset or clear the `/octomap_point_cloud_centers`. And when there is a `rtabmap_ros` loop closure the `/octomap_point_cloud_centers` does not clear and the resulting map is a combination of the corrected map due to the loop closure and the not corrected one.

wxmerkt, 2020-02-13:
Sorry I do not have the capacity or time to provide support to user questions. Based on checking the source code, `publishAll` should be called so the point cloud should be published (`publishPointCloud=true`) with an empty point cloud message. Did you check that there is nothing when you listen to the topic upon a reset?

acpl00, 2020-02-14:
I have tried the reset call service  and the the `/occupied_cells_vis_array` is reset, however  the `/octomap_point_cloud_centers` is not reset.

wxmerkt, 2020-02-14:
You may have to work with the source code and check that topics are published on reset (`publishAll`) and what those messages contain (e.g. `rostopic echo`)

wxmerkt, 2021-08-07:
Related to #85 - also due to the early return inside `publishAll` in
https://github.com/OctoMap/octomap_mapping/blob/84633da1077335e86f41fe214085d90dc212fcd8/octomap_server/src/OctomapServer.cpp#L487-L490

However, not yet fixed by #92