OctoMap/octomap_mapping #69: not showing entire octomap

borcherd, 2020-05-07:
I'm working on a project where a UAV needs to map an entire outdoor environment. Running in gazebo, i've got rtabmap which creates a pointcloud, and its being converted to an octomap with octomap_server to have the ability to save it to a .bt file. The only problem is that when looking at the octomap that octomap_server creates, its not the entire pointcloud converted to an octomap but only a part. 
[image]
my launchfile looks like this but I dont think that its got anything to do with that. 
<launch>
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
		<param name="resolution" value="0.2" />
		
		
		<param name="frame_id" type="string" value="map" />
		
		
		<param name="sensor_model/max_range" value="20.0" />
		
		
		<remap from="cloud_in" to="/rtabmap/scan_map" />
		<param name="base_frame_id" value="base_link" />
	</node>
</launch>

jiayinglin-rwth, 2020-07-02:
I also have the similar problem. It seems that the sensor_model/max_range is based on the fixed map frame, instead of the moving sensor frame(base_link). Which means if you set the max_range to 200 or more, or set the origin of the map frame in the middle of the octomap, you can see more of the octomap, but it is not a clean solution.

tadteo, 2020-09-22:
But they suggest to set the input in the map frame and the sensor_model in this way seems to be in the centre of the map frame. How can we correct this problem? Should we just modify the sensor_model/max_range or have I bad understand ho to fed the input to the algorithm?

jiayinglin-rwth, 2020-09-22:
Hey I have solved the problem. I had used the transformed pointcloud as input, which was actually wrong.
here is how we solved it:
1. input point cloud for the octomap need to be the one with the local coordinate (in our case it is odom, without any coordinate transformation)
please make sure that the header of the point cloud includes the right frame id
2. then do the coordinate transformation seperately (we implemented another node which use GNSS and IMU to do the transformation)
3. then in the octomap the fix map frame is "world" in our case, and base frame is the moving local frame
    <param name="frame_id" type="string" value="world" />
    <param name="base_frame_id" type="string" value="odom" />