OctoMap/octomap_mapping #70: Changes found in empty space

gstavrinos, 2020-05-12:
Hello,

I have been experimenting with the octomap tracking server and I encounter a weird problem. After mapping the environment and run the tracking server, I get a pointcloud that includes empty areas above the robot, like in the attached screenshot. (mapped area in white and occupied changes in red)

[image]

Any ideas why?

wxmerkt, 2020-05-12:
Hi George,
It's very hard to debug frames/pointclouds without insight into the processing chain (e.g. filters) and the tf transform trees, launch files, etc. Could you provide a reproduction case or describe these in more detail?

Thanks,
Wolfgang

gstavrinos, 2020-05-12:
Hey Wolfgang,

Thank you for getting back so fast.

I have a custom chain and I was suspecting that this was the issue, but I also tried the default octomap tracking server along with everything set at default and had the same problem. 

With a bare-bones reproducible process, I am not using any filters and have only set ground filtering to true for both mapping and change tracking.

Regarding the tf tree, it looks normal and roswtf does not report anything suspicious.
[image]

To sum up, using the default launch files (only changes include ground filtering and the pointcloud topic) I get the behaviour shown in the screenshot of the first post.

It has to be noted that about a week or two ago, I did not have this problem. Reverting back to previous commits of octomap_mapping and octomap_ros did not help. I don't really know if this is helpful, since the commits do not change anything related. Maybe a sync broke something, but I can't really think of anything.

Thanks in advance,

George

wxmerkt, 2020-05-12:
Hi George,
Thanks for the additional detail. Unfortunately it is really hard for me to debug or imagine what could be going wrong without a way to reproduce. We had a recent sync of Octomap to 1.9.5 (changes only related to build system and octovis moving to Qt5), as well as octomap_mapping from version 0.6.4 to 0.6.5 (adding color nodelet). Given that you've already tried to compile 0.6.4 of octomap_mapping locally, I am not sure what to suggest apart from working with the filter parameters and check/output the number of points in the pointcloud at every stage.

- Wolfgang

gstavrinos, 2020-05-12:
Hello again,

Thanks for your effort. I will try to debug it and come back, hopefully with a solution.

George

gstavrinos, 2020-05-14:
Ok, I have something, but it is not a solution. 

I have access to two Ubuntu 18.04 computers and one MXLinux 18 (Debian) computer. All of them have ROS Melodic installed. Both of the Ubuntu computers had exactly the same behaviour but the Debian-based one kept running properly... 

This might have to do something with the slower repo updates of Debian distros. I tried to pinpoint which library made the difference but could not find which. All of the computers are running PCL 1.8.0, octomap 1.9.0, octomap_ros 0.4.0 and octomap_mapping 0.6.4. (I purposely downgraded some packages to match my debian one)

Any tips would be greatly appreciated.

George