OctoMap/octomap_mapping #71: Performance issue

Jrdevil-Wang, 2020-06-10:
Hello, I am doing some experiments using octomap, and something strange occur on performance.

When I am using the default octomap version along with ROS melodic (Ubuntu 18.04), I get the debug log like this:

> Pointcloud insertion in OctomapServer done (0+307200 pts (ground/nonground), 0.564149 sec)

But when I use the version compiled with github kinetic-devel, it is much more slow and I get the debug log like this:

> Pointcloud insertion in OctomapServer done (0+307200 pts (ground/nonground), 6.756059 sec)

Any idea what would cause this difference? Many thanks!

wxmerkt, 2020-06-10:
How did you compile/what are your workspace settings? Is Release mode set?

Jrdevil-Wang, 2020-06-10:
Thank you for your quick reply!
It was my mistake that I forget to set release mode. Did not realize that there could be that much difference.
I recompiled the workspace with "catkin_make -DCMAKE_BUILD_TYPE=Release", and the performance looks fine.
Let me close this issue and sorry for your trouble.

closed 2020-06-10 by Jrdevil-Wang