OctoMap/octomap_mapping #72: After downloading the octomap_mapping file to the catkin_ws / src file directory, I get the following error when I make catkin_make in the catkin_ws file directory. Can you help me?

1617rosengineer, 2020-06-19:
-- +++ processing catkin package: 'octomap_server'
-- ==> add_subdirectory(octomap_mapping-kinetic-devel/octomap_server)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find octomap_ros (missing: octomap_ros_DIR)
-- Could not find the required component 'octomap_ros'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "octomap_ros" with
  any of the following names:

    octomap_rosConfig.cmake
    octomap_ros-config.cmake

  Add the installation prefix of "octomap_ros" to CMAKE_PREFIX_PATH or set
  "octomap_ros_DIR" to a directory containing one of the above files.  If
  "octomap_ros" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  octomap_mapping-kinetic-devel/octomap_server/CMakeLists.txt:20 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/cansel/catkin_wss/build/CMakeFiles/CMakeOutput.log".
See also "/home/cansel/catkin_wss/build/CMakeFiles/CMakeError.log".
Makefile:8972: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

wxmerkt, 2020-06-19:
As the message says, you need to install the `octomap_ros` package. The easiest way to do this is `sudo apt install ros-$ROS_DISTRO-octomap-ros`

1617rosengineer, 2020-06-19:
After downloading the file you said, when I made catkin_make again, I got the following error.

-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /home/cansel/catkin_wss/devel/share/octomap_msgs/cmake/octomap_msgsConfig.cmake:113 (message):
  Project 'octomap_msgs' specifies
  '/home/cansel/catkin_wss/src/octomap_msgs/include' as an include dir, which
  is not found.  It does neither exist as an absolute directory nor in
  '/home/cansel/catkin_wss/src/octomap_msgs//home/cansel/catkin_wss/src/octomap_msgs/include'.
  Check the issue tracker 'https://github.com/OctoMap/octomap_msgs/issues'
  and consider creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/octomap_ros/cmake/octomap_rosConfig.cmake:197 (find_package)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  octomap_mapping-kinetic-devel/octomap_server/CMakeLists.txt:20 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/cansel/catkin_wss/build/CMakeFiles/CMakeOutput.log".
See also "/home/cansel/catkin_wss/build/CMakeFiles/CMakeError.log".
Makefile:8972: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

1617rosengineer, 2020-06-19:
Thank you so much @wxmerkt

wxmerkt, 2020-06-19:
No, I have never used that particular package.

I will close this issue as the original question has been resolved.

closed 2020-06-19 by wxmerkt