OctoMap/octomap_mapping #73: Octomap map_server not saving the map

iandresolares, 2020-06-23:
Hello, I'm using octomap package in Kinetic distro in combination with the gmapping package in order to perform 2D SLAM and build a 2D map and also a 3D map with Octomap. Both are working fine, the 3D map is being created correctly in rviz but when I try to save it i get this error:

Request to /octomap_full failed; trying again...

In order to save it I'm using the command:

rosrun octomap_server octomap_saver -f name_of_map.bt

or

rosrun octomap_server octomap_saver -f name_of_map.ot

I get the same errors with the two different extensions. Any idea of what could be wrong here?

Thanks.

Andrés

wxmerkt, 2020-06-23:
Hi Andrés, 
Could you please provide the output of `rosservice list`? I have the suspicion this could be a namespacing issue.

Thanks,
Wolfgang

iandresolares, 2020-06-24:
Hi Wolfgang,
thanks for your answer. This is the complete output of the rosservice list:
bo/apply_joint_effort
/gazebo/clear_body_wrenches
/gazebo/clear_joint_forces
/gazebo/delete_light
/gazebo/delete_model
/gazebo/get_joint_properties
/gazebo/get_light_properties
/gazebo/get_link_properties
/gazebo/get_link_state
/gazebo/get_loggers
/gazebo/get_model_properties
/gazebo/get_model_state
/gazebo/get_physics_properties
/gazebo/get_world_properties
/gazebo/pause_physics
/gazebo/reset_simulation
/gazebo/reset_world
/gazebo/set_joint_properties
/gazebo/set_light_properties
/gazebo/set_link_properties
/gazebo/set_link_state
/gazebo/set_logger_level
/gazebo/set_model_configuration
/gazebo/set_model_state
/gazebo/set_parameters
/gazebo/set_physics_properties
/gazebo/spawn_sdf_model
/gazebo/spawn_urdf_model
/gazebo/unpause_physics
/gazebo_gui/get_loggers
/gazebo_gui/set_logger_level
/gazebo_gui/set_parameters
/robot/controller_manager/list_controller_types
/robot/controller_manager/list_controllers
/robot/controller_manager/load_controller
/robot/controller_manager/reload_controller_libraries
/robot/controller_manager/switch_controller
/robot/controller_manager/unload_controller
/robot/controller_spawner/get_loggers
/robot/controller_spawner/set_logger_level
/robot/controller_spawner_arm/get_loggers
/robot/controller_spawner_arm/set_logger_level
/robot/dynamic_map
/robot/front_rgbd_camera/rgb/image_raw/compressed/set_parameters
/robot/front_rgbd_camera/rgb/image_raw/compressedDepth/set_parameters
/robot/front_rgbd_camera/rgb/image_raw/theora/set_parameters
/robot/front_rgbd_camera/set_camera_info
/robot/front_rgbd_camera/set_parameters
/robot/imu/data/accel/set_parameters
/robot/imu/data/rate/set_parameters
/robot/imu/data/set_accel_bias
/robot/imu/data/set_rate_bias
/robot/imu/data/yaw/set_parameters
/robot/laserscan_multi_merger/get_loggers
/robot/laserscan_multi_merger/set_logger_level
/robot/laserscan_multi_merger/set_parameters
/robot/octomap_binary
/robot/octomap_full
/robot/octomap_server/clear_bbx
/robot/octomap_server/get_loggers
/robot/octomap_server/reset
/robot/octomap_server/set_logger_level
/robot/octomap_server/set_parameters
/robot/robot_imuimu/service
/robot/robot_state_publisher/get_loggers
/robot/robot_state_publisher/set_logger_level
/robot/slam_gmapping/get_loggers
/robot/slam_gmapping/set_logger_level
/robot/twist_mux/get_loggers
/robot/twist_mux/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/rviz/get_loggers
/rviz/reload_shaders
/rviz/set_logger_level

wxmerkt, 2020-06-24:
Thanks - this is as suspected due to a namespace clash: The `octomap_saver` node wants to request `/octomap_full` but your `octomap_server` is in namespace `robot`. As thus the octomap service is `/robot/octomap_full`. As a result, you can run the octomap saver node as (using remapping):

```
rosrun octomap_server octomap_saver -f name_of_map.bt octomap_full:=/robot/octomap_full
```

iandresolares, 2020-06-24:
Hello Wolfgang,
that was the problem, yep, it works great now! Thank you so much for the fast replies!!
I'm closing this issue.
Best regards,
Andrés

closed 2020-06-24 by iandresolares