OctoMap/octomap_mapping #76: any ROS2 port?

chopin1998, 2020-09-04:


wxmerkt, 2020-09-14:
Hi @chopin1998, @marwan99,
In reference to this post and https://github.com/ANYbotics/grid_map/issues/219#issuecomment-692288072: No plans or timeline for working on a ROS2 port from my side. Contributions are welcome (happy to create a `ros2` branch etc.), however, as I am currently still working on ROS1 I would not have any substantive feedback or time to devote to it.

Marwan99, 2020-09-17:
I can try and have a go at it. I propose splitting the port into the following PRs for ease of reviewing and to allow anyone interested in helping to jump in:
- octomap_server library & octomap_server_color library and updating CI
- octomap_server_node & octomap_color_server_node
- octomap_server_static
- octomap_server_multilayer
- octomap_saver
- octomap_tracking_server_node
- octomap_server_nodelet and octomap_color_server_nodelet

@wxmerkt What do you think? Please let me know if I missed something.

chopin1998, 2020-09-17:
hi, i found a port here

￼￼https://github.com/iKrishneel/octomap_server2

wxmerkt, 2020-10-05:
@Marwan99 your plan is clear and sounds good. I think it makes sense to have a maintainer for the ROS2 version - either in this repository or in a new repository in this organization. What do you think @ahornung?
I am happy to continue to look after ROS1, but currently do not yet work on ROS2 (starting later this autumn).

Thanks for the link to the port @chopin1998 - I did not look closer into it but it appears to be based on this repository.

iKrishneel, 2021-04-16:
@wxmerkt @ahornung 
I ported this repo to ROS2 ages ago. 
I am wondering if there is any chance to move the ros2 port to this organization?

iKrishneel, 2021-04-16:
@wxmerkt Thanks for your reply. As you suggested, I will update the headers and comments in the port.

ahornung, 2021-04-21:
@iKrishneel sure, that would be good! How different is the port, would it make sense to have it within this repo as separate branch or is it too different? Ideally, you could merge it back as a pull request - or what do you think @wxmerkt ?

iKrishneel, 2021-04-25:
@ahornung 
Since ROS1 support has ended, how about a separate repo for ROS2 version of octomap?

wxmerkt, 2021-04-25:
ROS1 is still supported until 2025 and widely used both in research and industry. Given that the code has diverged a bit, a separate repository might be the best option. Alternatively we could separate the communication and mapping parts and this would allow a single repository.

iKrishneel, 2021-04-25:
@wxmerkt I just updated the communication.

wxmerkt, 2021-12-24:
Thanks to the work of @wep21, there is now a ROS2 port on branch `ros2`.

closed 2021-12-24 by wxmerkt

wxmerkt, 2021-12-24:
@nuclearsandwich It appears I don't have access to the ROS2 release repository for `octomap_mapping` yet (it's not part of the team - https://github.com/orgs/ros2-gbp/teams/octomap/repositories). Many thanks in advance and happy holidays :-).

nuclearsandwich, 2022-01-05:
@wxmerkt apologies for the delay. There was an issue with the earlier deployment of changes which has now been resolved. Please confirm access to the https://github.com/ros2-gbp/octomap_mapping-release repository and let me know if there are any other issues.

Additionally, you're currently the only member of the release team. Should there be any other maintainers on it currently?

wxmerkt, 2022-01-05:
Thank you - releases for foxy, galactic, and rolling are out now (i.e. access is working :-)). Let's add @ahornung as the creator of the library, perhaps he has additional suggestions as well.

nuclearsandwich, 2022-01-07:
> Let's add @ahornung as the creator of the library, perhaps he has additional suggestions as well.

That change should be deployed or deploying soon.

wxmerkt, 2022-01-07:
Fantastic, thank you!

PR #95 (2021-12-24): ROS2 Migration
Closes #76