OctoMap/octomap_mapping #78: Installation problem for package "ros-noetic-octomap-server" on Debian 10

kbrix2000, 2020-12-08:
### Summary 

The installation of the Debian package "ros-noetic-octomap-server" fails on Debian 10 (buster), although according to [ROS Noetic installation page](http://wiki.ros.org/noetic/Installation) Debian 10 is supported.

### Required info

* Operating System: Debian 10
* ROS Version: noetic

### Reason

ros-noetic-octomap-server depends on libpcl-*1.8, but Debian 10 only provides libpcl-*1.9, see [Debian 10 packages webpage](https://packages.debian.org/search?suite=buster&searchon=names&keywords=libpcl). Debian 9 has offered libpcl-*1.8 packages, see [Debian 9 packages webpage](https://packages.debian.org/search?suite=stretch&searchon=names&keywords=libpcl).

Is it correct that the package depends on libpcl-*1.8 and libpcl-*1.9 at the same time (see package information below)?

### Error message

$ sudo apt install ros-noetic-octomap-server
Reading package lists... Done
Building dependency tree      
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-noetic-octomap-server : Depends: libpcl-apps1.8 but it is not installable
                             Depends: libpcl-common1.8 but it is not installable
                             Depends: libpcl-features1.8 but it is not installable
                             Depends: libpcl-filters1.8 but it is not installable
                             Depends: libpcl-io1.8 but it is not installable
                             Depends: libpcl-kdtree1.8 but it is not installable
                             Depends: libpcl-keypoints1.8 but it is not installable
                             Depends: libpcl-ml1.8 but it is not installable
                             Depends: libpcl-octree1.8 but it is not installable
                             Depends: libpcl-outofcore1.8 but it is not installable
                             Depends: libpcl-people1.8 but it is not installable
                             Depends: libpcl-recognition1.8 but it is not installable
                             Depends: libpcl-registration1.8 but it is not installable
                             Depends: libpcl-sample-consensus1.8 but it is not installable
                             Depends: libpcl-search1.8 but it is not installable
                             Depends: libpcl-segmentation1.8 but it is not installable
                             Depends: libpcl-stereo1.8 but it is not installable
                             Depends: libpcl-surface1.8 but it is not installable
                             Depends: libpcl-tracking1.8 but it is not installable
                             Depends: libpcl-visualization1.8 but it is not installable
E: Unable to correct problems, you have held broken packages.

### Package information:

$ apt-cache show ros-noetic-octomap-server
Package: ros-noetic-octomap-server
Version: 0.6.5-1buster.20201017.025706
Architecture: amd64
Maintainer: Wolfgang Merkt <opensource@wolfgangmerkt.com>
Installed-Size: 2284
Depends: libboost-atomic1.67.0, libboost-chrono1.67.0, libboost-date-time1.67.0, libboost-filesystem1.67.0, libboost-iostreams1.67.0, libboost-program-options1.67.0, libboost-regex1.67.0 (>= 1.67.0-10), libboost-serialization1.67.0, libboost-system1.67.0, libboost-thread1.67.0, libc6 (>= 2.14), libconsole-bridge0.4, libexpat1 (>= 2.0.1), libflann1.9, libfreetype6 (>= 2.2.1), libgcc1 (>= 1:3.0), libgl2ps1.4, libhdf5-openmpi-103 (>= 1.8.13), libjpeg62-turbo (>= 1.3.1), libjsoncpp1 (>= 1.7.4), liblog4cxx10v5, liblz4-1 (>= 0.0~r113), libnetcdf-c++4 (>= 4.2), libnetcdf13 (>= 3.6.1), libogg0 (>= 1.0rc3), libopenmpi3, libpcl-common1.9, libpcl-features1.9, libpcl-filters1.9, libpcl-io1.9, libpcl-kdtree1.9, libpcl-ml1.9, libpcl-octree1.9, libpcl-sample-consensus1.9, libpcl-search1.9, libpcl-segmentation1.9, libpcl-surface1.9, libpng16-16 (>= 1.6.2-1), libpocofoundation60, libproj13 (>= 4.8.0), libpython3.7 (>= 3.7.0), libqhull7, libsqlite3-0 (>= 3.5.9), libstdc++6 (>= 5.2), libsz2, libtheora0 (>= 1.0), libtiff5 (>= 4.0.3), libtinyxml2-6a (>= 5.0.0), libuuid1 (>= 2.16), libvtk7.1, libvtk7.1-qt, libxml2 (>= 2.6.27), ros-noetic-octomap, zlib1g (>= 1:1.1.4), libpcl-apps1.8, libpcl-common1.8, libpcl-features1.8, libpcl-filters1.8, libpcl-io1.8, libpcl-kdtree1.8, libpcl-keypoints1.8, libpcl-ml1.8, libpcl-octree1.8, libpcl-outofcore1.8, libpcl-people1.8, libpcl-recognition1.8, libpcl-registration1.8, libpcl-sample-consensus1.8, libpcl-search1.8, libpcl-segmentation1.8, libpcl-stereo1.8, libpcl-surface1.8, libpcl-tracking1.8, libpcl-visualization1.8, ros-noetic-dynamic-reconfigure, ros-noetic-nav-msgs, ros-noetic-nodelet, ros-noetic-octomap-msgs, ros-noetic-octomap-ros, ros-noetic-pcl-conversions, ros-noetic-pcl-ros, ros-noetic-roscpp, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-std-srvs, ros-noetic-visualization-msgs
Homepage: http://www.ros.org/wiki/octomap_server
Priority: optional
Section: misc
Filename: pool/ros/r/ros-noetic-octomap-server/ros-noetic-octomap-server_0.6.5-1buster.20201017.025706_amd64.deb
Size: 346544
SHA256: 4d831bbcc1ceaaa83f7c8853a2af140d2b6442f505c423501b581187aa1faea2
SHA1: 336f93be5304b34235ba163016d3aa85c0a17a89
MD5sum: e11445a8c0bcd7924f410e2b83621dec
Description: octomap_server loads a 3D map (as Octree-based OctoMap) and distributes it to other nodes in a compact binary format.
 It also allows to incrementally build 3D OctoMaps, and provides map saving in the node octomap_saver.
Description-md5: 4f44fb236321b686fc6abc0c3d293cab

wxmerkt, 2020-12-08:
Hi @kbrix2000,
Thanks for the report. That is definitely unintended. I tried tracking down which `rosdep` key would pull in v1.8 but could not immediately. I've removed our runtime dependency on `libpcl-all` such that it gets resolved by the upstream `pcl_ros` package instead. Fingers crossed this fixes it (but it will require a re-release to the buildfarm).

kbrix2000, 2020-12-15:
@wxmerkt Thank you very much for the correction! Did the build succeed? Unfortunately, I could not find an updated package on the [ROS repository](http://packages.ros.org/ros/ubuntu/pool/main/r/ros-noetic-octomap-server/), yet. I do not know ROS's update process for Debian packages. Is there still some kind of approvement required? Is it possible to download the updated package for a test somewhere else?

wxmerkt, 2020-12-15:
The package has not yet been updated due to a hold for a sync (the packages are built on a staging server ("testing") first and then are periodicially synced to the main PPA. The status is here: https://github.com/ros/rosdistro/pull/27679

kbrix2000, 2020-12-17:
@wxmerkt Thank you for the information. Would it be possible to download the updated package from the staging server and test it? What is the planned date for the next release?

wxmerkt, 2020-12-17:
You can change your ppa to testing by editing the `/etc/apt/sources.list.d/ros.list` from main to testing. apt-get update, and apt-get install should get you the bleeding edge package

kbrix2000, 2020-12-22:
What is the correct entry for `/etc/apt/sources.list.d/ros.list`? Unfortunately, `deb http://packages.ros.org/ros/ubuntu buster testing` does not seem to work. I also could not localize the updated Debian packages on the server.

wxmerkt, 2020-12-22:
The testing repository is described here - http://wiki.ros.org/TestingRepository - the URL needs to be:

```
deb http://packages.ros.org/ros-testing/ubuntu buster main
```

wxmerkt, 2021-08-07:
@kbrix2000 I hope the release in December resolved this - hence closing the issue now. Please reopen or let us know if the issue still persists

closed 2021-08-07 by wxmerkt