OctoMap/octomap_mapping #82: [octomap_server] Filtering In Map Frame

mohdomama, 2021-01-19:
Hi,
With  `filter_ground` as `false`, the filtering done with `pointcloud_min_z` and `pointcloud_max_z` happens in the map frame. I think the desired behavior would be to do this filtering in `base_link` frame.

Note: When `filter_ground` is `true`, all filtering happens in `base_link` frame.  

https://github.com/OctoMap/octomap_mapping/blob/bd66bc1f024f6d77ec2ad175d15c82746063c301/octomap_server/src/OctomapServer.cpp#L330

ahornung, 2021-01-21:
I think there are valid use cases for both scenarios. Ideally, there is a parameter for the filter frame.

mohdomama, 2021-01-22:
Hi,
Thanks for the reply. Should I add a parameter for this?

Also, when doing filtering with `filter_ground` as `false`, the  points outside the filter are not cleared. This issue is apparent when we have dynamic objects in the scene. The entire trail of the object comes as occupied. 
I addressed this issue in the PR by marking the points outside the filter as `pc_ground` which then get cleared in `insertScan` function.  Is this approach acceptable or should there be some other workaround for this?