OctoMap/octomap_mapping #84: fullmap displaying in RVIZ #Question

NicoNorena, 2021-03-08:
I create a custom node like the ColorOctree and I can publish  the binary msg and display it on RVIZ,
But It looks like i can't visualize the map when set as full map. Does the full map publishes with colors and node variables? in order to parse out the data from the node in the RVIZ plug in.

I get three huge boxes which does correspond to the occupancy space like the binary map. Can the octree from the full map be visualize like the binary map but also be able to retrieve the node data information ?

NicoNorena, 2021-03-11:
octomap_msgs::fullMsgToMap(*msg)

can I type cast to the child class?

NicoNorena, 2021-03-11:
This is the issue I see after serializing:
ERROR: Could not create octree of type "{CUSTOM_NAME}OcTree", not in store in classIDMapping

I did add getTreeType() in the  "{CUSTOM_NAME}OcTree".

wxmerkt, 2021-03-11:
Did you register your custom octree type with `registerTreeType(mytree)`, cf. https://github.com/OctoMap/octomap/blob/7a9ab4a357860f7263769cce9dd45107ec8b2c4a/octomap/src/AbstractOcTree.cpp#L205-L207?

NicoNorena, 2021-03-11:
@wxmerkt yes, in the header file the same way as in the ColorOctree in the StaticMemberInitializer class.
https://github.com/OctoMap/octomap/blob/devel/octomap/include/octomap/ColorOcTree.h#L187

Any other ideas?

NicoNorena, 2021-03-11:
I got it working!!
I had to a new instantiate "{CUSTOM_NAME}OcTree" pointer called tree

if(tree)
{
        std::stringstream datastream;
        if (msg->data.size() > 0)
        {
	        datastream.write((const char*) &(msg->data[0]), msg->data.size());
	        tree->readData(datastream);
        }
    }

Basically called the `octomap_msgs::fullMsgToMap(msg)`  directly on my plugin

closed 2021-03-11 by NicoNorena