OctoMap/octomap_mapping #85: /reset service does not work

saharLeisiazar, 2021-03-23:
Hi
I am using octomap_server on  ros Melodic and in my project I need to reset the /projected_map while mapping. But the /reset service does not work. I remember I had Kinetic before and did not have this problem but now that I am using Melodic, this service is not working. Is there any other way I can reset my map?

wxmerkt, 2021-03-23:
HI @saharLeisiazar,
Can you describe what happens when you call the reset service? Is there a response? The relevant code did not change between the released version for Kinetic (0.6.4) and the one released for Melodic (0.6.5) - cf. https://github.com/OctoMap/octomap_mapping/compare/0.6.4...0.6.5

saharLeisiazar, 2021-03-23:
Nothing happens. no response, no error
[image]

wxmerkt, 2021-08-02:
@saharLeisiazar @ValentinaMiller can you please try #92 and let me know if that fixes your issues?

PR #92 (2022-04-19): Fix publishing of projected map on reset
This should fix #85
closed 2022-04-19 by wxmerkt