OctoMap/octomap_mapping #86: process has died, exit code -11 Segmentation fault

Manit-Chan, 2021-04-05:
Once I have used this great library octomap_mapping in hector_quadrotor with Velodyne VLP16 project to show Marker array on Rviz and it work perfectly. 

But when i go to another project and need to apt full-upgrade, our hector_quadrotor project cann't work again.

We remove and reinstall ros and its dependencies after that hector_quadrotor project work again but octomap_mapping not.

When we use"roslaunch octomap_server octomap_mapping.launch" it say octomap_server: "process has died, exit code -11".

With gdb, we get more detail of error 

Thread 1 "octomap_server_" received signal SIGSEGV, Segmentation fault

sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory

This is octomap_mapping.launch

`<launch>
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server" >
		<param name="resolution" value="0.05" />
		
		
		<param name="frame_id" type="string" value="world" />
		
		
		<param name="sensor_model/max_range" value="5.0" />
		
		
		
	
	</node>
</launch>`

(point cloud topic (sensor_msgs/PointCloud2) is already published on /cloud_in so we dont need to remap)

Our environment

- Ubuntu 20
- Ros noetic
- PCL 1.10

I tried to search about "Segmentation fault" and it tend to be something wrong with PCL.

wxmerkt, 2021-04-05:
Are you using the Velodyne Gazebo plugin by any chance? We recently saw the error above coming from the Pointcloud-from-message conversion in another package and it was due to malformed LidarScans of the Velodyne Gazebo plugin. In this case compiling the latest Velodyne Gazebo plugin from source fixed it.

Manit-Chan, 2021-04-12:
Thank you wxmerkt  After i install Velodyne Gazebo plugin from source, It work again and now Marker array are shown on Rviz.

wxmerkt, 2021-04-12:
Great, thanks for the feedback. I'll go ahead and will close the issue.

closed 2021-04-12 by wxmerkt