OctoMap/octomap_mapping #87: Unable to launch any package

tomkimsour, 2021-04-26:
I'm new to the whole ros environment, did the tutorial and i need to understand how octomap works.
To do so i wanted to test your project.
I created a new workspace.
```
cd src
git clone https://github.com/OctoMap/octomap_mapping.git
cd ..
catkin_make
source devel/setup.bash
```
From there everything went fine, it compiled without a problem.
Then I wanted to launch a file but i got : `[octomap_mapping] is not a launch file name
The traceback for the exception was written to the log file
`

Could you help me with this ?

wxmerkt, 2021-04-26:
The error message says that what you tried to launch isn't a launch file - as thus could you please provide the command you used? The following should work:
```
roslaunch octomap_server octomap_mapping.launch
```

tomkimsour, 2021-04-26:
> The error message says that what you tried to launch isn't a launch file - as thus could you please provide the command you used? The following should work:
> 
> ```
> roslaunch octomap_server octomap_mapping.launch
> ```

It was it : i did ```roslaunch octomap_mapping``` instead of ```roslaunch octomap_server octomap_mapping.launch```

Thanks a lot

closed 2021-04-26 by tomkimsour