OctoMap/octomap_mapping #89: No ground plane found in scan

tomkimsour, 2021-05-25:
Hello,

I am using ros kinetic and am actually trying to do mapping with octomap but I can't understand properly what this error mean.
It only prompt when i set ground_filter to true.

```
[pcl::SampleConsensusModelPlane::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[ INFO] [1621410313.723860512]: PCL segmentation did not find any plane.
[ WARN] [1621410313.723899495]: No ground plane found in scan
```

I have no idea where it could be coming from. My odometry seems right, i can map using my lasers.

here is my lauch file : 

```
<?xml version="1.0"?>

<launch>
    <!-- Depth Image -> Pointcloud -->
    <node pkg="nodelet" type="nodelet" args="manager" name="depth_image_nodelet_manager" output="screen"/>
    <node pkg="nodelet" type="nodelet" name="pepper_pointcloud" args="load depth_image_proc/point_cloud_xyz depth_image_nodelet_manager --no-bond">
      <remap from="camera_info" to="/naoqi_driver/camera/depth/camera_info"/>
      <remap from="image_rect" to="/naoqi_driver/camera/depth/image_raw"/>
    </node>

    <!-- Octomap server  -->
	<node name="octomap_server" pkg="octomap_server" type="octomap_server_node" output="screen">
    <remap from="cloud_in" to="points" />
    <param name="frame_id" type="string" value="/odom"/>
    <param name="base_frame_id" type="string" value="/base_link" />
    <param name="height_map" type="bool" value="true"/>
    <param name="sensor_model/max_range" value="-1"/>
    <param name="sensor_model/hit" value="0.7" />
    <param name="sensor_model/miss" value="0.4" />
    <param name="sensor_model/min" value="0.12" />
    <param name="sensor_model/max" value="0.97" />
    <param name="latch" value="false" />
    <param name="filter_ground" value="true" />
    <param name="resolution" value="0.05" />
    <param name="occupancy_max_z" value="1.6" />
	</node> 

    <node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen" >
      <remap from="/scan" to="/naoqi_driver/laser"/> 
      <param name="map_update_interval" value="2" />
      <param name="linearUpdate" value="0.01" />
      <param name="angularUpdate" value="0.01" />
      <param name="temporalUpdate" value="0.5" />
      <param name="xmin" value="-10." />
      <param name="ymin" value="-10." />
      <param name="xmax" value="10." />
      <param name="ymax" value="10." />
      <param name="particles" value="300" />
      <param name="maxRange" value="2.5" />
      <param name="maxUrange" value="2.0" />
      <param name="delta" value="0.02" />
    </node>

  <node pkg="move_base" type="move_base" respawn="false" name="move_base_node" output="screen">
    <rosparam file="$(find navigation)/nav_config/move_base_params.yaml" command="load" />
    <rosparam file="$(find navigation)/nav_config/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find navigation)/nav_config/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find navigation)/nav_config/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find navigation)/nav_config/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find navigation)/nav_config/dwa_local_planner_params.yaml" command="load" />
    <rosparam file="$(find navigation)/nav_config/navfn_global_planner_params.yaml" command="load" />
  </node>

	<node pkg="rviz" type="rviz" name="rviz" args="-d $(find navigation)/rviz_config/octo_config.rviz">
	</node>

<node pkg="rostopic" type="rostopic" name="arbitrary_name" args="pub -1 /joint_angles naoqi_bridge_msgs/JointAnglesWithSpeed -- '[1, now, Head]' '[HeadPitch]' '[-0.2]' 0.05 0" output="screen"/>

</launch>
```

UdayRockzz, 2021-12-10:
Hi @tomkimsour  Did you solve this issue?

tomkimsour, 2021-12-13:
> Hi @tomkimsour Did you solve this issue?

As far as I remember it was due to the robot lense coefficient but I never solved this problem.

HuaYuXiao, 2024-03-12:
I guess it's because your lidar is 2d lidar rather than 3d lidar.