OctoMap/octomap_mapping #90: octomap_server pointcloud_[min|max_z] and occupancy_[min|max]_z parameters only allow values up to -/+100

ndugdale, 2021-07-27:
I have been trying to use the octomap_server to generate an OctoMap that goes up to +/- 120 from the map origin. Points outside the +/-100 range are excluded from the OctoMap, despite setting larger parameter values for pointcloud_[min|max]_z in the launch file.

My launch file looks like this.
``` xml
<launch>
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
		<param name="resolution" value="0.1" />
		<!-- fixed map frame (set to 'map' if SLAM or localization running!) -->
		<param name="frame_id" type="string" value="/map" />
		<!-- maximum range to integrate (speedup!) -->
		<param name="sensor_model/max_range" value="30.0" />
		<param name="latch" value="false"/>
                 <param name="pointcloud_min_z" value="-200.0"/>
                 <param name="pointcloud_max_z" value="200.0"/>
		<!-- data source to integrate (PointCloud2) -->
		<remap from="cloud_in" to="/cloud" />
	</node>
	<node pkg="mapping" type="tf_broadcaster.py" name="tf_broadcaster" output="screen"></node>
        <node pkg="mapping" type="nav_publisher.py" name="nav_publisher" output="screen"></node>
</launch>
```

I can see that my launch file is correctly accessing the parameters, because if I instead set the pointcloud_[min|max]_z to have a magnitude less than -/+100 (e.g.  -/+20), these values will be reflected in the terminal with,

``` shell
~/catkin_ws$ rosparam get /octomap_server/pointcloud_min_z
-20.0
~/catkin_ws$ rosparam get /octomap_server/pointcloud_max_z
20.0
```

However, if the parameter is set to be greater than -/+100, it appears to be reset to -/+100. Is this the intended function? The package summary [here](http://wiki.ros.org/octomap_server) suggests that the intended default value for these parameters are -/+ infinity. 

Thanks!

aladin2022, 2022-03-23:
Hello, is the pointcloud_min/max_z reflects the limit height of the points cloud to be projected in the occypancy grid map ?

laelliott, 2023-06-16:
Hopefully you aren't still stuck on this but in case someone else is.

It's happening because the dynamic rqt config is boxing the cloud and map it to -100 to 100 

If you look at the config you can see where it is set 

https://github.com/OctoMap/octomap_mapping/blob/6e96c831795c664ba554bdff53844657a7916f83/octomap_server/cfg/OctomapServer.cfg#L12