OctoMap/octomap_mapping #93: How to integrate into existing package?

ValentinaMiller, 2021-08-05:
I am currently working with this workspace https://github.com/mSimon12/multiple_quadrotors and I am trying to track the changes in the octomap and therefore would like to use your packages. However, I am not able to connect the two packages.

When I try to start the any of the octomap_server launch files it tells me "Nothing to publish, octree is empty" even though I can see the octomap in rviz, can save the octree, use its services....

Am I missing something?

Thanks!

wxmerkt, 2021-08-05:
`Nothing to publish, octree is empty` indicates that the Octree is empty and thus the server won't publish any message. Publishing is either triggered after point cloud insertion, clearing parts of the octree, or calling the reset service. As thus, I am surprised that you can see the Octomap if it indicates that the Octree is empty. The first step to investigate is to check that the topics are connected properly. `rqt_graph` (which may not work on Noetic?) would be the first step, or alternatively you can use `rosnode info [name of node]` and explore the topics with `rostopic list`.

ValentinaMiller, 2021-08-06:
Thanks for your quick response!  I am working with melodic but in the rqt_graph everything seems to be connected as it's supposed to be. I wonder whether I do something wrong while launching the nodes. Is there anything special I have to take care of?
Thanks!

wxmerkt, 2021-08-07:
The relevant code is in [OctomapServer.cpp](https://github.com/OctoMap/octomap_mapping/blob/kinetic-devel/octomap_server/src/OctomapServer.cpp) for the regular octomap server or in [TrackingOctomapServer.cpp] for the tracking server - based on the post I am not sure which one is used or you are planning to use. From the linked repository, it appears that the standard octomap server is used fusing information in the `camera/depth/points` topic (cf. [here](https://github.com/mSimon12/multiple_quadrotors/blob/main/src/quadrotor/slam/launch/octomap_mapping.launch)) - this likely creates the Octomap you can visualise (along with the raw point cloud that is the basis of this octomap. Note, this is the standard octomap server and not the tracking octomap server - i.e., it wont publish changes but only the full octomap

ValentinaMiller, 2021-08-09:
Thanks again! Using camera/depth/points used part of the issue I think, the octomap_talker subscribed to camera/depth/points now and the octomap_tracking_server subscribed to octomap_talker and octomap_listener subscribed to octomap_tracking_server. (I guess that is how it is supposed to be?)
However when I am moving and listening to the octomap_tracking_server nothing is being published...
Do you have any idea why?