OctoMap/octomap_mapping #97: Hi, points queue? issue

Minsu-Kim-KNU, 2022-01-14:
Hi, I'm trying to mapping with LIVOX-MID-70 LiDAR.

In my case, firstly, points are stacked few moments and then, stacking is stop.
I edited .launch file in world frame as camera_init, and point cloud's topics to.

any solutions?

wxmerkt, 2022-01-14:
Hi @Minsu-Kim-KNU 
There isn't enough information to help us debug - e.g. if the topics are correctly mapped and the transforms are received soon enough/at all. It's also possible that the node is getting quite busy integrating at a high rate and updates are slow (e.g. CPU at 100% for `octomap_server` but RViz not updating etc.). It's easier to debug if you have a rosbag and corresponding used launch files; but overall - it's very hard to say anything based on the information provided.

Minsu-Kim-KNU, 2022-01-14:
Hi, @wxmerkt 
Thanks for your reply.
Sorry for my few informations, and sorry for my poor English. 
I tried to solve this problem myself, but all the informations below are my current situation.
1. Run `octomap_server octomap_mapping.launch` in Indoor condition, It works well. I can get all the informations that I want.
2. But outdoor Condition, I cannot. I run this launchfile with LIVOX MID-70 LiDAR and with 'livox_mapping mapping_mid.launch`.
livox_mapping is work well. very well. but, octomap_mapping isn't. It means, It's update is too slow, and it cannot get all the points from LiDAR. After mapping , I got too sparse pointclouds with octomap_mapping. It means, few number of points.

I explained my situation as best I can. Sorry for no result Image, sorry for my English.

wxmerkt, 2022-01-14:
I am afraid I cant provide much insight here - it is possible that the sensor/max_range is clipping too soon. It's also unclear if the received data is too much. If it's working indoors, the issue is likely to be with the settings in the launch file that need to be adjusted for your use case.

Minsu-Kim-KNU, 2022-01-16:
Thanks for your advice. I'll do my best with this issue, and if I have an updates I'll mention in here. thanks.

closed 2022-01-16 by wxmerkt

Minsu-Kim-KNU, 2022-01-21:
Hi, I solved this problem 3days ago.

My problem was usage of CPU. Mapping with 2 launch files is too hard for my CPU.
so, octomap was not upadate well. Then I tried mapping with only octomap, it works well. thanks for your advice.