OctoMap/octomap_mapping #103: octomap_saver_node report an error

ShuiYunXi, 2022-05-09:
Hi,I'm trying to use the octomap_saver_node to save a map,but report an error as follows:

environment:Ubuntu20.04
ROS's Version:foxy
Question 1:
```
ros2 run  octomap_server octomap_saver_node
[INFO] [1652079216.465871727] [octomap_saver]: Map received (0 nodes, 0.020000 m res), saving to 
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 18446744073709551613) > this->size() (which is 0
)
```

Question 2: The octree map will move with the movement of the robot, can you tell me how to solve it? thank you very much

note:The transform include the odom->base_link and have no  map.

wxmerkt, 2022-05-09:
This is due to an underflow bug when the `map_name` is not provided (parameter `octomap_path`): 
https://github.com/OctoMap/octomap_mapping/blob/7b7e8e87207241462a531445afed51ef1b637014/octomap_server/src/octomap_saver.cpp#L101

@wep21 - could you please take a look at this / check for length / throw error?

ShuiYunXi, 2022-05-09:
how can i check  this / check for length / throw error?

wep21, 2022-05-12:
@wxmerkt Thank you for letting me know. I will take a look at this issue.

nkr-seasony, 2022-06-22:
Hi I am having  the same problem , by  running ros2 run octomap_server octomap_saver_node -f map.bt ,I get this:
[INFO] [1655901876.946400726] [octomap_saver]: Map received (23542 nodes, 0.050000 m res), saving to 
terminate called after throwing an instance of 'std::out_of_range'
 what():  basic_string::substr: __pos (which is 18446744073709551613) > this->size() (which is 0)
How should I fix it?

wep21, 2022-06-22:
@nkr-seasony I added instruction to save octomap in #108. Please check it out.

nkr-seasony, 2022-06-24:
Thank you, I fix it.