OctoMap/octomap_mapping #111: octomap_mapping

WhoElse554, 2022-12-04:
Hi,
i am very new to octomap and want to know if there is a way to load an already existing .bt file in rviz using octomap?
I tried to publish my .bt to the octomap_server and tried to subscribe to the server, but when i launch the octomap_mapping.launch file, i am getting the error "Nothing to publish, octree is empty". Can someone tell me what the approach is for existing binary files and what i need to subscribe/publish to get it vizualised in RVIZ, please?
Thanks in advance!

alvaropascuaI, 2022-12-13:
Hi @WhoElse554 , has been able to solve?, I have the same problem but i cant resolver

ssaigarimella, 2025-02-26:
@WhoElse554 @alvaropascuaI same issue here, were you able to solve?

alvaropascuaI, 2025-03-05:
I don't remember how far I got or if I even solved it. I'm sorry

lydiaplu, 2025-05-02:
I used the command ros2 run octomap_server octomap_server_node --ros-args -p octomap_path:=/home/my_octomap_map.bt -p frame_id:=map -p resolution:=0.05 to load the .bt map. I checked the source code in octomap_server.cpp (located at octomap_mapping/octomap_server/src/octomap_server.cpp) and found that around line 332, the code loads the file path from the octomap_path parameter.
In RViz, I configured MarkerArray (topic: /occupied_cells_vis_array) and PointCloud2 (topic: /octomap_point_cloud_centers) to visualize the 3D Octomap.