OctoMap/octomap_mapping #115: Warning : Nothing to publish, octree is empty

MallakAbunimeh, 2023-02-06:
I am using cyglidar to get the 3D pointcloud to generate a 3D map in rviz using the octomap_server package. The following 2 images shows the cyglidar.launch file & the octomap_server.launch file  

Cyglidar launch file:
```
<launch>
  <!-- Arguments -->
  <arg name="version" default="2" doc="version type [0, 1, 2]"/> <!-- 0: 2D, 1: 3D, 2: Dual -->
  <arg name="frequency_level" default="0" doc="level [0 to 15]"/>
  <arg name="pulse_control" default="0" doc="pulse mode [0, 1]"/> <!-- 0: Auto, 1: Manual -->
  <arg name="pulse_duration" default="10000" doc="duration [0 to 10000]"/>

  <node pkg="cyglidar_d1" type="cyglidar_pcl_publisher" name="line_laser" output="screen">
    <param name="port" value="/dev/cyglidar"/>
    <param name="baud_rate" value="3000000"/>
    <param name="frame_id" value="laser_link"/>
    <param name="fixed_frame" value="/map"/>
    <param name="run_mode" value="1"/>
    <param name="frequency" value="$(arg frequency_level)"/>
    <param name="set_auto_duration" value="$(arg pulse_control)"/>
    <param name="duration" value="$(arg pulse_duration)"/>
  </node>

  <node pkg="rviz" type="rviz" name="rviz" args="-d $(find cyglidar_d1)/rviz/cyglidar_config.rviz"/>
  <node pkg="tf" type="static_transform_publisher" name="to_laserlink" args="0 0 0 0 0 0 1 map laser_link 10"/>
</launch>

```
octomap_server.launch file:
```
<!-- 
  Example launch file for octomap_server mapping: 
  Listens to incoming PointCloud2 data and incrementally builds an octomap. 
  The data is sent out in different representations. 

  Copy this file into your workspace and adjust as needed, see
  www.ros.org/wiki/octomap_server for details  
-->
<launch>
	<node pkg="octomap_server" type="octomap_server_node" name="octomap_server">
		<param name="resolution" value="0.022" />
		<!--param name="height_map" value="false" /-->

		<!-- fixed map frame (set to 'map' if SLAM or localization running!) -->
		<param name="frame_id" type="string" value="map" />
		
		<!-- maximum range to integrate (speedup!) -->
		<param name="sensor_model/max_range" value="-1" />

		<!-- minimum and mzximum height of points -->
		<param name="pointcloud_min_z" value="0.02" />
		<param name="pointcloud_maz_z" value="1.0" />
		
		<!-- data source to integrate (PointCloud2) -->
		<remap from="cloud_in" to="scan" />
	
	</node>
</launch>

```

After launching the cyglidar.launch file then the octomap_server.launch file I got this error. I also added the MarkerArray option in rviz and changed the topic to /occupied_cells_vis_array. Then, I got the following error:
[ WARN] [1675673929.993534530]: Nothing to publish, octree is empty.

I tried to verify the pointcloud is being published to scan topic but I get nothing back. I am using Ubuntu 20.0.4 and ros noetic version.

wxmerkt, 2023-02-06:
Can you check the `rqt_graph` and potentially turn on DEBUG for the rosconsole (e.g. with `rqt_console`). It might be that messages are being ignored or topics arent connected

MallakAbunimeh, 2023-02-06:
Hello, thanks for the reply. This is what the output was for `rqt_console`:
[image]
And this is the output for the `rqt_graph` with every parameter not hidden:
[image]

wxmerkt, 2023-02-06:
`rqt_graph` shows that your connections are not connected - `octomap_server` expects `scan` whereas your driver publishes `scan_3D`

MallakAbunimeh, 2023-02-07:
@wxmerkt 

We changed the topic from scan to scan_3D in the octomap_mapping.launch file. However, we still get the same warning message and the topics published by the octomap_server are showing up as dead sinks. 
[image]

The following image is also the topic list when the server is running:
[image]

We're not sure what exactly we're supposed to change to make sure the topics are connected.

wxmerkt, 2023-02-07:
They are showing as dead sinks since no node is subscribed to them - and that includes RViz. Please try to activate the DEBUG output in the console settings to see more detail on why messages may have been ignored. Also make sure the frames / transforms required you are mapping in are correct and available

MallakAbunimeh, 2023-02-12:
I tried to launch the cyglidar.launch file again and then the octomap_mapping.launch file with RVIZ instead of launching it in the cyglidar package and then got this error:
[image]

[image]

After activating the DEBUG output in rqt_console I got this:
[image]

And here are the connections on rqt graph:
[image]

MallakAbunimeh, 2023-02-12:
I tried to run the octomap_mapping launch file again and got this error:
[image]

Here is the rqt graph output:
[image]

The DEBUG output:
[image]

[image]

[image]

wxmerkt, 2023-02-12:
Do the transforms from `map` to your lidar frame exist?

MallakAbunimeh, 2023-02-14:
It only shows when I launch rviz in cyglidar.launch file:
[image]

wxmerkt, 2023-02-15:
Are they connected? cf. rqt_tf_tree or echo the transform

MallakAbunimeh, 2023-02-16:
This is what I got:
[image]

Note: . This only appears when I launch RVIZ in the cyglidar.launch file; however, after commenting the RVIZ launching node in the cyglidar.launch file, I was unable to open RVIZ in the octomap_mapping.launch file. So, can you also check line 27 for me please? I want to know why RVIZ is not opening in the octomap_mapping.launch file. Is this line correctly written? Or I need to rewrite it or do something to make it launch? Because I want to get the 3D map in the octomap_mapping launch file.

Line 27 in octomap_mapping launch file is shown below:
[image]

MallakAbunimeh, 2023-02-16:
[image]