OctoMap/octomap_mapping #120: ros2 launch octomap_server octomap_mapping.launch.xml failed

tejalbarnwal, 2023-05-24:
Hello,
I am working with ros2 foxy. I have a topic which publishes point cloud. When I try doing `ros2 launch octomap_server octomap_mapping.launch.xml` , I get the following error.

[image]

I am not sure what the error related to undefined symbol refers to. Could you please help me?

tejalbarnwal, 2023-05-24:
Also, I used `https://github.com/ros-perception/perception_pcl` with `foxy-devel` branch

wxmerkt, 2023-05-26:
@wep21 Could you assist here please?

wep21, 2023-05-28:
@tejalbarnwal I'm sorry, but I cannot reproduce the error with octomap_server installed via apt in docker.
Did you also build some libraries from source in your slam workspace?