OctoMap/octomap_mapping #123: Reading a .bt or .ot file using the octomap_server_node not working properly

NadimDeeb, 2023-10-24:
I've been using the octomap_server to visualize the octomap of an area and after the processing and publishing is done I use the octomap_saver using this command: `ros2 run octomap_server octomap_saver_node --ros-args -p octomap_path:=map.bt` to save the map. When I try and re-display it in RViz using the `ros2 run octomap_server octomap_server_node map.bt` I get a message that that the octree is empty.
I tried making multiple .bt files all along the process as well as at the end but all are the same. I've noticed that the files are a couple of MB big and their contents are the following:
```
# Octomap OcTree binary file
# (feel free to add / change comments, but leave the first line as it is!)
#
id OcTree
size 14640642
res 0.05
data
��#�#�#�#�#�#�#�#�#�#��# ....
```
 Am I doing anything wrong?

yunusstalha, 2024-04-09:
You should try ros2 run `octomap_server octomap_server_node "map.bt"`.
In ros noetic using  `rosrun octomap_server octomap_server_node _file_path:=map.ot `(or.bt) does not work, saying that octree is empty but 
` rosrun octomap_server octomap_server_node "map.ot" (or .bt)`, works without problem.

closed 2024-05-25 by wxmerkt

ssaigarimella, 2025-02-26:
@yunusstalha @NadimDeeb does the .bt file have to be in a specific path? This still is not working

lydiaplu, 2025-05-02:
I used the command ros2 run octomap_server octomap_server_node --ros-args -p octomap_path:=/home/my_octomap_map.bt -p frame_id:=map -p resolution:=0.05 to load the .bt map. I checked the source code in octomap_server.cpp (located at octomap_mapping/octomap_server/src/octomap_server.cpp) and found that around line 332, the code loads the file path from the octomap_path parameter.
In RViz, I configured MarkerArray (topic: /occupied_cells_vis_array) and PointCloud2 (topic: /octomap_point_cloud_centers) to visualize the 3D Octomap.