OctoMap/octomap_mapping #133: How to use in ROS2

vcozzolino, 2024-06-10:
Hi,

I have installed succesfully all the packages on ROS2 Humble with Ubuntu 22.04.

Now I'm trying to figure out how to run the map creation. I tried looking online but I can't find much information. Assuming that I have Gazebo open with a robot deployed in it, what do I have to launch to start the data collection from the point cloud? Which scripts do I have to invoke?

Tejal-19, 2024-11-12:
Is the issue solved ?

gritelli, 2024-11-20:
On Humble, I'm trying to generate an octomap using the following command:
`ros2 run octomap_server octomap_server_node --ros-args -r cloud_in:=/amr_1/pcd_map -p frame_id:=map`
but I'm getting this :
```
[INFO] [1732100713.304798138] [octomap_server]: Publishing latched (single publish will take longer, all topics are prepared)
[WARN] [1732100713.310499430] [octomap_server]: Nothing to publish, octree is empty
[WARN] [1732100713.310569819] [octomap_server]: Could not open file 
[INFO] [1732100713.822978716] [octomap_server]: Message Filter dropping message: frame 'odom' at time 633.900 for reason 'discarding message because the queue is full'
[INFO] [1732100713.923510084] [octomap_server]: Message Filter dropping message: frame 'odom' at time 634.000 for reason 'discarding message because the queue is full'
...
```
I'm using a 3D lidar SLAM that publishes a pointcoloud2 topic (`/amr_1/pcd_map`). Not really sure how to address this.

kalhansb, 2025-02-09:
> On Humble, I'm trying to generate an octomap using the following command: `ros2 run octomap_server octomap_server_node --ros-args -r cloud_in:=/amr_1/pcd_map -p frame_id:=map` but I'm getting this [...]

Cannot say for sure, you are probably missing tf for 'odom'. In your case the tf tree should be something like:
map -> odom -> base_link -> amr_1's frame.
Provide the tf tree using tf2_tools view_frames for further debugging.